daid/LegacyCura #1202: Nozzle temperature for Prusa i3 not set

net-id, 2015-04-01:
I've been using 14.09 on my Prusa I3 without a problem.
I then tried 15.02 but found that the target temperature for the nozzle is not set on the printer when I try to print. It stays at 0. Bed target temperature is set correctly. It's like Cura is sending it the wrong gcode.

daid, 2015-04-01:
Nothing changed on this area, except that the bed is heated first by default now, so the hotend does not sit still while being hot for too long.

net-id, 2015-04-01:
Thanks for the speedy response.
Ohhhh. Right... I just tested this and now I see that it leaves the nozzel turned off till the bed reaches it temperature first. Got it. I thought there was something drastically wrong :)

Having the nozzle warm to half way or a user settable value might save some waiting time - just a suggestion. 

Thanks             ...Laurie:{)

closed 2015-04-02 by nallath