daid/LegacyCura #1215: Developer Doc and Support - How to modify and compile Cura

mo22de, 2015-04-13:
Hey,
may this has been asked but I really can't find something usefull on it. And may it is offtopic here but I hoped Daid would answer directly to me on that!? Can we talk Daid?

Well the instructions from the repo readme helped me but did not lead to some other answeres which come along.

May someone could help me out on this:
- adding the python requests lib to the Cura Build routine for all platforms
- compile Cura for several platforms

I'm just able to compile on a 64bit Ubuntu for debian_amd64 and on a 32bit ubuntu to compile for debian_i386.

Win7 64 bit fails to compile via git command as recommended ending up following the error description in a situation where it wants a "usr/shared" folder on my windows system.

While compiling for debian and installing the request library was missing, but actually there wasnt a python a all on that machine so Cura just dont start. And after installing python it just hangs. Starting from Console it let me know that the requests library wasn't available.

So I guess there's a way to automate that?
And is there a way to add the requerst library to the windows portable python as well?

Yes I'm new to python and this kind of compiling routine for multiple platforms. But I'm not a beginner at all in programming, so a hint could may be all I need. So pls help me getting to this.

THX

daid, 2015-04-14:
Did you install the packages from:
https://github.com/daid/Cura#debian-and-ubuntu-linux

mo22de, 2015-04-14:
Thanks for getting to me daid.
Yes I used that instruction and repo, the branch git creates is SteamEngine.

Well may I didnt mention clear enough that I'm on that Code adding some stuff and compiling on Ubuntu in a VM for debian_amd64 nicely.

And hell yeah, I learned a lot last weeks spending on this, but the compiling and espacially the windows build routine just makes me crazy ^^
Is it right that dependency management to include request is the right way to go?

But how to get it for windows python portable and how to include it?

mo22de, 2015-04-14:
Well reading my own words now I myself didn't really get my questions ;-)
So let me try to ask more precisely:

* Adding a python library to cura and compile routine
  * What steps where to do?
    * package.sh ?
    * build_app.py ?
    * requirements ?
    * /scripts/linux/debian_control  [Depends: ... python-requests] add this for request lib?
  * However will the Linux User need to run 'sudo apt-get install python-requests' anyway?
    Meaning that this needs to be done today too for a cura user not developer with this:
    'sudo apt-get install python-opengl python-numpy python-serial python-setuptools python-wxgtk2.8 curl'
    Or is this automated by installing Cura, and if yes then how to do with another package like python-requests?

* Compile Environment for build target Windows 
  * What is a working setup on a Win7 64Bit Machine?
  * python V ?
  * git V
  * other tools needed, like MinGW and what Version of it??
What are possible Commands an parameters that work 'git package.sh win32' respectivly 'git package.sh win32 -z' ? Those didn't worked for me on Win7 64 bit.
And not on a VM Ubuntu 64 bit and not on a Ubuntu 32 bit netbook running:
'./package.sh win32'

* For other builds, is there more to know about adding a python library, like for ARM build and can those all be run from a Ubuntu 64 Bit Machine?
'./package.sh debian_armhf'
'./package.sh darwin'
'./package.sh freebsd'
'./package.sh fedora'

And just for completion what about RaspberryPi and BeagelBoneBlack?
Has somone compiled and run it on a BeagleBone Black ?
Would the debian_armhf target such a device?

So this is a lot of questioning and I'm not sure if I need to know everything about it. I just need to add this one library, may someone could help me out doing that if that is faster then explaining it to me!? ;-)

daid, 2015-04-14:
For a complete release build, only the package.sh script is needed. So that should be the entry point.

Note that we are currently rebuilding the UI from scratch, which is one of the reasons that I'm not working on updating proper dev documentation.

mo22de, 2015-04-14:
Thats great to hear as I was intended to do the same while adding some features to it first.
As I said I'm new to python so this was my first lesson with it ;-)

Are you still going to use python?
Will you make use of a gui framework?

Any chance to get a look at your plans or may get involved in it?

I would like to add my features to cura, which was planed as a mod build release, may you like it and want to make it part of the origin!?

Anyway I will finish it up and may we could have a chat then and I will show you what I'm doing to your cura right now?

Thanks for your time so far! :-)
Have a great time and keep the good work and espacially your accessibility!