daid/LegacyCura #1216: Build plate chessboard size

dotBits, 2015-04-14:
As you can see on the right side of the screenshot, seems that the build plate chessboard is not wide enough.

[image]

mo22de, 2015-04-14:
I just wonder what happend to your settings menu, shouldn't be there more options?
Like printing Temperature
and Filament diameter an Flow?

At least thats what my Cura shows me. :-)

Cant imagine that the Ultimaker 2 Extended didn'T have the option to change that?

dotBits, 2015-04-14:
this is a fresh install on Windows 7 and the configured machine is an Ultimaker2

mo22de, 2015-04-14:
It's the same on Ubuntu with Ultimaker2 at my Computer.
And my fault, it seems like Ultimaker has removed it since Ultimaker2 to the machines controller. :-/
So you have to click wheel those settings at the machine while switching filament.o_O

In my opinion a bad idea, but may you could setup a new machine in Cura with Ultimaker2 Settings as a custom machine.

Would be interesting what would happen. Also interesting would be why they do that, the only reason may would be variable temperatures and speeds during one print, but even that could be achieved using a setting that comes with the GCode.

I dont want to capture this issue-post, but anyone has the answer to that too.
The original qustion and/or Problem was the build plate chessboard size not matching the build plate.

daid, 2015-04-14:
It matches the build volume just fine, the nozzle (which is on the left of the head) cannot reach the right area of the buildplate. (Just check it in your machine, you'll notice that there is still room on the buildplate when the head can no longer move to the right)

The older reprisentation of the build volume was actually wrong, it was the same size, but centered, meaning you lost a bit more volume on the virtual clips.

closed 2015-04-14 by daid

dotBits, 2015-04-14:
@daid clear that it's just a visual issue for the chessboard and not for the build volume itself, what about to wrap the available printing area into a build plate container and draw the chessboard on its size?