daid/LegacyCura #1219: Travel speed on first layer(s)

Gregwar, 2015-04-17:
I'm printing bottom layers really slowly to secure the print start, and I think It should be great to have a parameter available to tune the travel speed (not extrusion speed) on first layer.

Another thing is that, for instance, drawing a small circle "on the glass" is more likely to fail than drawing a simple line. Maybe there could be a correction on the speed that would depend on the shape curve the nozzle is following.

BagelOrb, 2015-04-17:
Please submit separate issues for separate ideas.

Why would you want different travel speeds in the first layer?
Why are you printing at lower speeds on the first layer(s)?

Gregwar, 2015-04-17:
I print slowly to make the filament stick the plate
I am printing PLA directly on glass at 60deg
Reducing extrusion speed (bottom layer speed) works great but the nozzle still travel fast, so the filament is ripped off
I can produce a video demostrate the problem

Dim3nsioneer, 2015-04-18:
I can confirm the strong dragging mentioned by @Gregwar especially on printed shell parts when the print head moves fast from one perimeter to the other. It happens with high travel speed which is a good thing in general due to reduced oozing.
However, a reduced travel speed on the first layer might help avoiding ripping off printed perimeters but it will lead to additional oozing with probably a majority of hotends. Reducing hardware controlled parameters such as acceleration or x/y-jerk might be another approach. They might also help for the second issue menitoned in the OP.

m-vo, 2015-04-19:
+1 for a reduced travel speed (option) on the first layer. I'd say: The smaller the outline, the bigger the effect of dragging and chance of ruining the whole thing. Imho especially for a first layer with lots of holes in it having more oozing would be the smaller problem. (?)

(The only stable solution I found for that problem was printing a 'solid' first layer and drilling the holes afterwards.)

BagelOrb, 2015-04-20:
Sounds legit!

MarkJB, 2015-07-21:
I've been seeing this too. With shell thickness set to 3x nozzle dia, printing inner holes, the outlines are printed 1 outline thick and the start and end of the shell don't meet and are dragged away as the print head goes of the do a single outline on another hole and the same thing happens. If it would print all the lines that make up the shell it would probably work ok.

BagelOrb, 2015-07-23:
@MarkJB This issue is about travel speed; not about the outer walls.

MarkJB, 2015-07-23:
@BagelOrb I should have said I dialled down my travel speed (from 120 to 60) to try and see what was happening and observed the above. The speed of the head moving between outer walls is what is ripping the print off the bed. But where only one perimeter is printed, it is far more likely to be ripped off the bed because of a fast travel move.