daid/LegacyCura #1220: software security questions

ryno-511, 2015-04-17:
Hi, I bought a printrbot for work and would like to use it on a networked computer.  The IT department will only allow me to install on a network computer if I can provide answers to a number of specific questions about Cura (printrbot's recommended software).  I hope this is an appropriate forum to ask these questions (if not, please advise).  
1) Does the app use the .NET framework & if so what version?
2) Does Cura install OpenSSL dlls (libeay32.dll and ssleay32.dll) & if so what versions?
3) Does Cura install Microsoft Visual C++ dlls & if so what version?
4) Does Cura install Java & if so what version?
5) Does Cura install Microsoft VB6 ActiveX controls & if so what version?
6) Does Cura install GDIPlus.dll for graphics support & if so what version?

By the way, I took the printrbot home lsat night, plugged it into my mac, and Cura worked perfectly.  Zero issues!  Now I hope I can do the same at work on the windows computers...

Thanks!

daid, 2015-04-18:
1) No.
2) Nope.
3) Yes, but not system global, it's installed in the application directory, version 9.0
4) Nope.
5) No no no.
6) Nope.

Note that the whole Cura application is contained within the install directory, except for the user configuration files which are installed in the users own directory (default C:\users\[name]\.cura)
So it's easy to review what comes with it, and what not. Also all the Code in Cura itself is open source and thus can be reviewed.

-Daid
Ultimaker R&D

closed 2015-04-18 by daid

ryno-511, 2015-04-18:
Above and beyond. Thanks so much!
-Ryan