daid/LegacyCura #1229: feature request: overextrusion for Infill

Chrisu02, 2015-04-28:
A feature to overextrude the infill a little bit would be helpfull when printing with very thin layers/nozzlewidth to make a part more sturdy.

BagelOrb, 2015-04-28:
Overextrusion is not something you want. The extruder might not be able to provide the given amount of overextrusion and start skipping. Try increasing the infill percentage.

nallath, 2015-05-01:
Well not quite true; Over extruding does increase the strength as the layers bond better. Increasing the infil percentage will make it stronger because of other reasons (more material)

Chrisu02, 2015-05-01:
Maybe i wrote it in a missunderstanding way, i meant only the infill of the inner part not the top/bottom infill!
as nallath mentioned it, it will strengen the part from within it, without giving really overextrusion issues to the top/bottom layers.

BagelOrb, 2015-05-03:
Overextrusion itself is a term for a problem, the 'over' part means 'too much', which is not something you ever want.
However, another way to conceptualize it is by saying you want a higher line width for the infill. Then you will also not run into problems when infill is at 100% and similar problems.

Though this might be a viable idea, I think it comes down to overextrusion for most 3D printers: taking a line width higher than the nozzle width will cause problems: the printer won't keep up with how much it has been told to extrude.

BagelOrb, 2015-05-03:
However, I have been thinking about using this scheme the other way around: for 'underextrusion'.

If we make the line width smaller than the nozzle size, e.g. for top/bottom skin, that might result in better intralayer bonding between the different lines within a layer and smoother skins.