daid/LegacyCura #1230: feature request: top layer different layerheight.

Chrisu02, 2015-04-28:
A feature to also change the Top layer to a smaller one would help printing the part itself faster with higher layers but still give the top a smoother surface, for example when printing badges with letters.

BagelOrb, 2015-04-28:
This would only be for the topmost layer, and not for all top skins.
The enhancement would only be usable for a minimal amount of models.

Allowing for a different layer height in the top skin would be fairly difficult.
Usually the skin is printed last of all layer parts, but when we would print a skin part at half the layer height, we cannot print it last anymore, since it should be half the layer height lower than the rest.

Dim3nsioneer, 2015-04-28:
IMHO, only a change to assigning different layer heights to different z ranges makes sense to work at. But that change has a bit a longer time range... However, everything else is just cosmetics. Just my two cts.

daid, 2015-04-30:
Actually, just changing the top layer thickness would result in worse prints. Because of the sudden change in pressure you get extrusion difference.

There is a long and difficult feature on our "complex feature" list which is about different layer heights in the model. Which is a really complex thing to achieve in the current code.

(Closing this, as the feature as proposed makes little sense for the amount of effort required in the Engine)

closed 2015-04-30 by daid