daid/LegacyCura #1236: Cura 15.02.1 takes 3 minutes before slicing begins. Bug?

MaccieBob, 2015-05-03:
Cura 14 and all previous versions begin slicing immediately after the "phasing in" of the .stl file

I am now using Cura 15.02.1 on default settings, but it takes about 3 or 4 minutes after the .stl "phases in" before the slicing process starts. 

This happens with all/every .stl file used: they all do the same thing, no matter how simple or complex, no matter if the .stl is my own generated file or a downloaded file. 

The model eventually slices. Once the model is moved or manipulated, another period of waiting is required before slicing again begins.

Windows 7 Home Premium 64 bit
Intel i7 
16Gb RAM
GeForce GTX 760
128Gb SSD

daid, 2015-05-03:
And virusscan or firewall software?

MaccieBob, 2015-05-03:
I have temporarily disabled (Windows) firewall and (Panda) antivirus software, but the result is the same.

No additional software was installed between the installation of the previous 14.12.1 and the current 15.02.1.

Cura version 14.12.1 is installed concurrently with 15.02.1. Cura 14.12.1 still works 100%.

An additional finding is that 15.02.1 hangs when trying to close the program. However, if the model has (eventually) sliced, then Cura 15.01.1 can be closed immediately.