daid/LegacyCura #1243: Strange G-Code creted when slicing OpenSCAD Model

JustAnother1, 2015-05-14:
I created a OPenSCAD Modell and exported an STl File. I then open that STL File using Cura 15.02.1 and the model looks good.
[image]

But if I look at the Layer view there seems to be a problem:
[image]

The Model is:
```openscad
$fn=50;

difference() {
	union(){
		// Wand Seite
		translate([-11, -8.9, 0.4]) rotate([40, 0, 0]) cube([20, 20, 1]);

		// LED Halter
		translate([-10, -8, -2]) cube([6, 15, 2.4]);
		translate([2, -8, -2]) cube([6, 15, 2.4]);

		// Knick
		translate([-11, -7.8, 0]) rotate([0, 90, 0]) cylinder(h=20, r=2);
	}
	// LED Streifen
	cube([200, 11, 1.5], center = true);
}
```

daid, 2015-05-14:
Settings used?
Version?

(Z hop is known to screw with the layer view sometimes)

JustAnother1, 2015-05-14:
Version is as stated 15.02.1
Settings are: 0.2mm Layer thickness, what else do you need? Where can I get them? I'm willing to change whatever helps,...
("Cool head lift" has no effect on the problem)

BagelOrb, 2015-05-15:
File > Reset profile to defaults > layer height = 0.2 > problem not reproducible

Note that you didn't really say what the problem is, but I can kind of infer the problem from the incredibly low-res image you posted ;)

Perhaps you should try the dev version of openSCAD, since that is the version I am using to 'compile' your code...

closed 2015-05-15 by BagelOrb