daid/LegacyCura #1244: Feature request: automatically react to physically impossible(?) prints.

juj, 2015-05-16:
I'm looking at this model in Cura: http://www.thingiverse.com/thing:235008

When viewing the layers of the model, I interpret that the tip of the scythe of the model should be physically impossible to print, since it's an example where the printer would have to deposit filament literally in thin air. See the following screenshot:

[image]

This kind of configuration looks like it should be possible to automatically detect and react to. I'm not sure what is the best/appropriate reaction, but possible ones would be something like:

   - don't attempt to print any volume that is physically impossible with the current settings. Those pieces of volume are discarded from the print job gcode, and the layers view would show those volumes drawn in solid gray color to mark that it's not possible. The gray color could also be shown in the Normal view mode to make it easy to spot.
   - print a warning that given the current print settings, the model contains pieces of volume that are impossible to print, and recommend that the user sets a print option that builds suitable support to print.
   - have an advanced mode checkbox that forces printing of volumes even though they have been deemed physically impossible. (for odd cases where users know what they want)

Sorry if this is a duplicate discussion (I'm somewhat new to the 3D printing world), and thanks for reading!

JustAnother1, 2015-05-16:
+1 I would also want to have this feature. As this is one thing I always need to check manually. Would be good if Cura would at least give a warning.

Dim3nsioneer, 2015-05-16:
For the time being you might be right with the physically impossible. However, such a structure could be printed by an SLS printer. There is not yet an open source SLS printer afaik but you never know what's up to come... ;-)
A warning might be useful. However, it might be quite tricky to detect overhangs starting in thin air.

JustAnother1, 2015-05-16:
@Dim3nsioneer To be future proof there could be a flag in the machine settings that tells cura if the printer is SLS or FDM.

It is not easy to do this, but should be doable. The code that calculates the support structures might be usable. That code probably has to solve a similar problem.

juj, 2015-05-16:
Pseudo-thinking a method, it feels like it would be possible to categorize undersides to "positive" and "negative" slopes by doing a "topological" flood fill of sorts on the voxels of the data, by starting from the bottom layer, performing a flood fill to layers upwards and marking connectivity between voxels. If one has to back down to a layer below during the search, then that indicates there is an underside with a negative slope. I'm not at all familiar with the code though, so that's just a handwavy thought, most likely the code does something more advanced already :)

XiaoShiZi, 2015-05-16:
Opensource sls ;-)
http://reprap.org/wiki/OpenSLS
Nice experiments on a lasercutter ;-)
Greetz, Xiao Shi Zi

Dim3nsioneer, 2015-05-17:
@juj CuraEngine does not work with voxels afaik. However it does register the number of objects per layer. If a subsequent layer has more 'objects' than the layer before, this might be an indication for a free hanging something. But it might also just be a body that splits up into two or more branches. But maybe standard overhang detection is good enough to find such spots as it represents a 90° overhang angle on the 'first' layer.

BagelOrb, 2015-05-18:
Indeed, doing a check only when the number of objects changes shouldn't be such a large hit on performance. I imagine the check would consist of subtracting the previous layer (with an outset) from the current layer and checking the area of the resulting parts. When area is larger than X, warn the user.

However, user warnings are not yet supported, so I will keep this issue open until user feedback from the engine is implemented.

juj, 2016-01-03:
Hey, checked again today, and still seeing this in Cura 15.04.3. Here is another example: trying to print file heartless-dragon.stl from http://www.thingiverse.com/thing:17309/#files with Ultimaker 2, and all Cura settings at default, first set High resolution in quickprint, and then switched to full settings to enable "Support type: Everywhere". However examining the print job in layers view, it looks like the software is attempting the impossible by printing in thin air (see the areas highlighted inside black ellipses):

[image]

Warning the user would be a good start, though in this case, I don't know what to do to get the model to print correctly, and tried the different setttings in the "Support type" menu, but I was not able to make Cura generate support to carry the whole left paw of the model. I wonder if there might exist settings that would work, or if it's a current limitation of Cura? On the Thingiverse page, I see 28 makes of the model, but I could not find if any of them would have been printed with Cura(+Ultimaker 2). Can someone verify if there exists Cura settings for the above model to print correctly without the left paw collapsing?

JustAnother1, 2016-01-03:
I know this model as example for two colour prints. In Two colour prints people seem to use two STL files (one for each extruder). That is presumable the reason this model of the dragon without the heart exists.

Printing in mid air with two nozzles is not a problem, as the heart then exists.

Then in the Instructions it clearly gives tips on how to print with only one nozzle. From you screen shot it looks possible to add support for the paw. So maybe a Cura only issue. Have you tried other slicers?

juj, 2016-01-03:
Did a test print of the model at normal quality settings in quickprint mode, and then switched to advanced mode to set "Support type: Everywhere". Here is a time lapse video of the print: https://dl.dropboxusercontent.com/u/40949268/dump/DragonHeartless.mp4 (89MB, 60 seconds). The print came out ok, except that indeed the left paw is partially missing support and as result one of the toes came out garbled.

This looks like a software bug, and it feels a bit like the "Support type: Everywhere" is being treated similar to "Support type: Touching buildplate" and Cura does not like to extend the support structure over the tail of the dragon even though "Support type: Everywhere" has been set? There is a small difference in these two settings in how much support is generated, but the support structure did not cover the tail in this case.

Actually, I'm not familiar with any other slicing software, and I've only been using Cura since it was paired as "the official" Ultimaker 2 printing software. Are there some worthwhile ones to try that are compatible with Ultimaker 2 on Windows?

BagelOrb, 2016-01-04:
The new Cura (Cura 2) will have solved this support problem.
You can check out a beta release at http://software.ultimaker.com/Cura_closed_beta/15.10/

juj, 2016-01-04:
Neat, downloaded `Cura-1.99.0-dev.20151117-win64.exe` and tested both the heartless dragon and the skeleton models there, and examining the layers view, the new version generates correct support for both models. Great work, I'm happy to call this closed as working in the next release.

Btw, I see the url says it is a closed beta - I noticed a few other things, is there a place I could report items on the new 1.99.0 dev version?

closed 2016-01-04 by juj

BagelOrb, 2016-01-04:
Yes there is!
It's https://github.com/Ultimaker/Cura