daid/LegacyCura #1245: Necessary support is not added even when Support Type: Everywhere is specified.

juj, 2015-05-16:
This is related to issue #1244, but slightly separate, so marking this as a separate discussion to keep the focus in both distinct.

When I open the skeleton model from #1244, and set Support Type: Everywhere, and take a look at the layers view, I see the following:

[image]

which looks to me like Cura did build support (in dark blue) for the raft and for the torso of the skeleton, but it decided to not add support for the (tip of the) scythe at all? Is that accurate? (I did not try printing because of this, but I can give it a go to verify) Cross-comparing to the screenshot in issue #1244 however, it seems clear that a FDM printer (I've set Cura to use Ultimaker 2) cannot print the tip of the scythe without adding support, so it looks like it would have to add support in this case. Does this look like a bug, or perhaps there is some advanced knob I haven't understood to configure this?

(I ended printing the model lying flat on its back, which worked ok, but thought that this deserved to be mentioned at least)

daid, 2015-05-17:
The latest internal beta does generate support for this case. We totally re-did the support code due to problems like this for the next major release.

closed 2015-05-17 by daid

juj, 2015-05-17:
Great to hear, thanks!