daid/LegacyCura #1249: Misprint in 1cm height with 100% fill setting(ABS)

QoSi, 2015-05-20:
Hello,

i can print with a lower setting fill with 95% there are no mistakes.
I print with the Ultimaker 2, in some forums the problem  is returned to a faulty heat insulator but logically considered it may, with the difference of the settings not be.

daid, 2015-05-20:
So, what do you want? Is there a problem with Cura or not?

QoSi, 2015-05-20:
I think that the problem lies in the software, because the problem is so is due to different setting.

sorry for my english, I hope you can understand the explanation

nophead, 2015-05-20:
It isn't possible to print 100% volume with ABS because it is too viscous to fill the sharp corners of a rectangular filament channel with a bead that starts out with rounded sides.

QoSi, 2015-05-20:
However, I need a 100% fill my object when air holes in the contact are deformed it in contact with boiling water.

the peint with 100% setting is below and above the 1cm height ok
in 1cm height, there is always a gap, the layers are not properly connected, it is too little rather than too much material.

nophead, 2015-05-20:
You would need an infinite amount of pressure to not have any air gaps at all. On most machines the nozzle is simply forced upwards. On the other hand PLA can be extruded in a very liquid state by raising the temperature so can achieve virtually 100% fill.