daid/LegacyCura #1252: Layer numbering inconsistent between g-code and GUI

Nesciosquid, 2015-05-22:
Cura generates comments for each layer in the g-code, which is great.

However, when troubleshooting, I often want to compare the g-code against the Layers view in the GUI. This can be confusing, since the Layers view is 1-indexed, but the g-code comments are 0-indexed, so what should be layer 1 in the g-code is actually layer 2 in the GUI, etc.

Is this intentional? I understand that perhaps starting from 0 in the GUI would be less intuitive for non-programmers, but the inconsistency is rather confusing unless you're looking for it.

daid, 2015-05-22:
It's unintentional, but as you already guessed. The layer numbering in the GCode comes from a programming brain, while the number in Cura comes from a user brain.

Nesciosquid, 2015-05-22:
Yep, that makes sense, but I'm still wondering whether reconciling the two would be worth it.

You could either adjust the g-code to be 1-indexed, which (probably) wouldn't cause any harm since Cura's layer comments seem to be arbitrary anyway, or you could change the UI to start at 0 instead of 1. The latter might confuse an uninformed user.

Maybe an option in the preferences setting to switch between the two?

daid, 2015-05-22:
Really? A preference setting? Everything configurable isn't always a good thing (look at eclipse as a fine example)

It would make sense to make the gcode 1 based as well. But I'm not sure if I will break someones plugin that way.

Nesciosquid, 2015-05-22:
That's a good point, re: unnecessary preferences! I don't think that's a good option.

"Breaking" someone's plugin is an interesting thing to be concerned about. It seems to me that the plugins that interact with layers are already 'broken' in a certain sense, when considering user input (That's how I realized this inconsistency existed.)

Take the case of the PauseAtZ plugin, which lets you enter either a height or a layer. Is that layer the number of the layer as it shows up in Cura, or the number as it shows up in the gcode? Should plugins be designed to compensate for Cura's 1-indexed labeling, or should they be faithful to the g-code instead?

Dim3nsioneer, 2015-05-22:
I can only speak for the TweakAtZ plugin: This one is consistent with the CuraEngine numbering as it is applied to the GCode generated by the Curaengine. And this one starts at 0.
I once had a look into the Cura code with the intention to issue a pull request which would bring the GUI numbering in agreement with the Curaengine numbering. I simply gave up because the +1 is hardcoded at so many places it would be almost impossible to catch them all.
But it's certainly a point which has to be addressed in PU.