daid/LegacyCura #1253: Can't install Cura 15.02.1 on Ubuntu 14.04 - dependency errors not satisfiable

jsquaredz, 2015-05-24:
Trying to install Cura on ubuntu 14.04.  Getting the following errors when trying to install:
cura depends on python-wxgtk2.8; however:
  Package python-wxgtk2.8 is not installed.
 cura depends on python-opengl; however:
  Package python-opengl is not installed.

This happens when I try to install the .deb from the ultimaker website.

daid, 2015-05-25:
Latest 15.04 release should fix the python-wxgtk issue. Not sure if that will resolve the python-opengl issue.

Ruff9, 2015-08-08:
Same issue here, just can't install Cura on Ubuntu 14.04. 

dpkg: error processing package cura (--install):
dependency problems - leaving unconfigured

Does anybody have a solution to this ?