daid/LegacyCura #1255: Cura with bfb printers

paweleczek, 2015-05-26:
Cura works well for printers BFB (one extruder). I'm using the printer BFB 3000.
Bookmarks preSwitchExtruder and postSwitchExtruder not work, why cura can not print on both extruders. In addition, BFB printer uses the M101 command - turn 1 extruder, a second extruder turn M201, M104 - the temperature setting 1 extruder, M204 - the temperature setting 2 extruder, for 1 extruder M108, M208 for 2 extruder. 
If the pre and post bookmarks work well, you can set the heating and cooling of heads, in addition cura engine should detect the witch extruder print. If the first is to use the commands M104, M101, M108, if the second extruder is a command M204, M201, M208.

Someone has already tried to do it, why it does not work?
In the file :
CuraEngine/src/gcodeExport.cpp

if (flavor == GCODE_FLAVOR_BFB)
    {
        //For Bits From Bytes machines, we need to handle this completely differently. As they do not use E values but RPM values.
        float fspeed = speed * 60;
        float rpm = extrusion_per_mm * speed * 60;
        const float mm_per_rpm = 4.0; //All BFB machines have 4mm per RPM extrusion.
        rpm /= mm_per_rpm;
        if (rpm > 0)
        {
            if (isRetracted)
            {
                if (currentSpeed != int(rpm * 10))
                {
                    //fprintf(f, "; %f e-per-mm %d mm-width %d mm/s\n", extrusion_per_mm, lineWidth, speed);
                    //fprintf(f, "M108 S%0.1f\r\n", rpm);
                    *output_stream << "M108 S" << std::setprecision(1) << rpm << "\r\n";
                    currentSpeed = int(rpm * 10);
                }
                //Add M101 or M201 to enable the proper extruder.
                *output_stream << "M" << int((extruderNr + 1) * 100 + 1) << "\r\n";
                isRetracted = false;

Thanks!!!

BagelOrb, 2015-05-28:
Dual extrusion is not fully supported yet. We will be working on it in the near future.

closed 2015-05-28 by BagelOrb

daid, 2015-05-29:
On the BFB side, Dual extrusion has never properly been tested. I did some "pre-work" for it, but most likely it needs more adjustments. We don't own these machines to test on, so we 100% depended on user feedback for BFB support.

paweleczek, 2015-05-29:
Thank you for your response.
I have a printer BFB 3000. Thank you for your support in Cura BFB printers . Cura gives better results than Axon and KISSlicer .
I can test the printer BFB . I can prepare files .bfb (two extruders) from Axon and KISSlicer . I know the commands gcode BFB printers . I think I know what to add in Cura , printing of two extruders , but I do not know how to do it. Many users BFB looking forward to it.
I want to help
Paweł