daid/LegacyCura #1256: model size  = memory limits ?

elipeo, 2015-05-27:
i'm trying to test slicing models for Building Xtreme 1000 machine (x 820 y 780 z 850).
The size of the model to slice is  x 407 y 407 z 805 : Cura

The model ( x 192.31 y 192.31 z 380.03 mm) is nicely sliced (layer 0.2) by Cura with .ini file for Big Dual Builder. The G-code file is +/- 263 Mo.
The model size is

elipeo, 2015-05-27:
Sorry for double post:
Cura is computing the big model almost at the end and stopped - the computing bar window disappear.
So how to break this limit.

daid, 2015-05-27:
Windows? On windows you hit some hard memory limits per process. If you are on Mac/Linux then these limits are not there and you can use up all the memory you want.

elipeo, 2015-05-27:
yes windows 7 with 64 Gb ram (°__°) ... I've just installed Cura & tested the big file (*__*) hyper-prima, it works like a charm : thanks a lot {I didn't know those memory limits per process}

elipeo, 2015-05-27:
on a Macbook pro with 8 gb ram.

closed 2015-05-27 by elipeo