daid/LegacyCura #1257: weight filament > Cura estimation

elipeo, 2015-05-27:
Have we take an up-margin for the estimation weight and filament footage. 
The purpose is to print a sculpture in one piece. Cura estimated the print at 8086 g' / 27011 m filament 1.75 mm. The dual system of Builder can manage 2 x 5 kg spool. So I have still an open limit for 2 kg.
It's a big challenge °+° : 664h of printing ...

BagelOrb, 2015-05-28:
I do not understand the problem.
What are you asking?

closed 2015-05-28 by BagelOrb

elipeo, 2015-05-29:
The probleme would be to compute an under-estimated weight for a very long printing. Let's say one month printing and 500 g' missing at the end. I've just tested a "short" print with a Builder, 42h30 of real time printing were estimated at 37h50 by Cura.
The weight was computed at 146 g' and is in real, 142 g'.
In this sense, it's reassuring.

daid, 2015-05-30:
Weight is estimated by density, you can configure the density of your material in the preferences.

elipeo, 2015-05-30:
Plus one :) thanks for this precious accuracy.

BagelOrb, 2015-05-31:
The time estimations are known to not be so accurate.
If you are willing to dig into it and enhance the code then go right ahead!

For now this has low priority for us.

daid, 2015-05-31:
Actually, the time estimates depend on the firmware settings:
https://github.com/Ultimaker/CuraEngine/blob/master/src/timeEstimate.cpp#L9

Which most likely don't match your machine.