daid/LegacyCura #1258: Feature Request (sort of): Set XYZ position on all moves

quillford, 2015-05-28:
Setting the position of every axis even if it hasn't changed would be helpful in the case that something like a homing or jogging is pressed mid print in the host. For example, if I pause a print in host X and host X allows me to jog while paused, and I jog an axis. Then I unpause the print and it is printing in the wrong  position because the axis I jogged was not set in the next line of gcode.

daid, 2015-05-29:
It would also mean more data storage is needed and more data transfer (when USB printing).

I would say, this is a host bug, as you should not be able to jog mid-print.

closed 2015-05-29 by daid

BagelOrb, 2015-05-31:
It is not a bug. Mid-print jogging is just an unsupported feature.
I would say the firmware should be able to backtrack until it finds the last move which does have a Z position, and use that one.

Note that after unpausing the next action shouldn't be the next line of gcode; it should first move to the xyz position of the current line of gcode before executing the next. Otherwise we could start extruding from the arbitrary position the head was in after pausing...