daid/LegacyCura #1263: Command line slicing errors and limited to single stl input

DefProc, 2015-05-31:
I'm trying to generate build plates of several files from a bash script to allow for semi-automated production of resized items. Specifically, I'm testing with these stls: https://github.com/cheapjack/buildyourown/tree/master/cad/hands/baylee using Cura 15.02.1 on Ubuntu 64bit.

If I run: 

`cura raptor_2.0_fingertip-{1..5}.stl` 

Cura opens as expected, with all 5 parts on the build plate, and produces gcode for all the items. However, if I run: 

`cura --slice --output=fingertips.gcode raptor_2.0_fingertip-{1..5}.stl`

cura gives the error: 

```
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/share/cura/Cura/util/sliceEngine.py", line 416, in _runEngine
    self._result.addReplaceTag('#P_TIME#', self._result.getPrintTime())
  File "/usr/share/cura/Cura/util/sliceEngine.py", line 93, in getPrintTime
    return _('%d minutes') % (int(self._printTimeSeconds / 60) % 60)
NameError: global name '_' is not defined

GCode file saved : fingertips.gcode
```

It does produce a gcode file, but it is only for the first item, and it identical to the output produed by running: 

`cura --slice --output=fingertip.gcode raptor_2.0_fingertip-1.stl`

which has the same error message. 

Should Cura be able to deal with for multiple input files from the command line?

BagelOrb, 2015-06-01:
Perhaps you should check Ultimaker/Cura
The new interface is not likely to have the same problem.

closed 2015-06-01 by BagelOrb

Mauroalfieri, 2015-07-02:
Hi BagelOrb,
   I have the same problem on Raspberry Pi kernel 3.18.5+ on armv6l

./cura.sh -i /home/pi/Cura/config/CuraConfig.ini -s staffaSharpL.stl -o staffaSharpL2.gcode
load preferences from /home/pi/.cura/dev/preferences.ini
Listening for engine communications on 49674
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 552, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 505, in run
    self.__target(*self.__args, **self.__kwargs)
  File "Cura/util/sliceEngine.py", line 416, in _runEngine
    self._result.addReplaceTag('#P_TIME#', self._result.getPrintTime())
  File "Cura/util/sliceEngine.py", line 93, in getPrintTime
    return _('%d minutes') % (int(self._printTimeSeconds / 60) % 60)
NameError: global name '_' is not defined

Thank's
Mauro