daid/LegacyCura #1264: Heated Bed Upgrade Problem 25,4mm anstatt 10mm

Makersteff, 2015-05-31:
Hi,i have Upgraded my Ultimaker with the Original Heated Bed Upgrade. The Problem is that the Z-Achs now make 25,4mm (1inch) but not 10mm what it must to do!
I think the new Z-Screw has a other spindle pitch as the Original that was in the machine.

Thanks for any Help
Steffen

Dim3nsioneer, 2015-05-31:
Adjust your machine settings in Cura (ticking the HBK) and upgrade to the newest firmware. Then it will do what it is supposed to do... ;-)

Makersteff, 2015-05-31:
(ticking the HBK)???
i have already Upgraded the Firmware.

Dim3nsioneer, 2015-05-31:
Sorry, my fault. You have to install a new printer (Ultimaker original). In the initial settings, select the heated bed kit (hbk). Then install the firmware again. It will take a different one with correct settings.

Makersteff, 2015-05-31:
Ok Thank you for Help ;-)