daid/LegacyCura #1273: Size of the drawing

eef051542, 2015-06-05:
Cura won't accept drawings that are a pretty good size but not exceeding the dimensions stated in the users manual. I have an Ultimaker 2 and the size of the part is 8" x 8" x 2"H.

Any suggestions?

Thanks,

Gene

daid, 2015-06-06:
8 barbaric units is about ~20.4mm. This is about the outer limits of the machine I think. But there is also space reserved for the skirt and the metal clips on the bed. First, you want to set the number of skirt lines to 0 (in the expert settings) this removes the extra space for the skirt and will allow you to use the space of the bed more optimal.

eef051542, 2015-06-06:
Thanks daid, that coincides with my observations.

closed 2015-06-06 by eef051542