daid/LegacyCura #1274: shell thickness for support different than the printed object (feature request ?)

francoisba, 2015-06-08:
Hi,
I noticed in the 15.04 version (mac) that updating shell thickness for the printed object, impact also the thickness of the support wall.

Is there any solution to set both thickness separately ?
It would save material ! ;)

Thank you for your help,
Best regards

[image]
[image]

BagelOrb, 2015-06-08:
Will look into it tomorrow

closed 2015-06-09 by BagelOrb

francoisba, 2015-06-09:
Whow ! Thank you for your responsiveness !
To be completely honest with you, I forgot to mention that at that moment, the «spiralize» option was checked..
Thank you for your help !
Best regards