daid/LegacyCura #1275: No infill in 15.04 version ?

francoisba, 2015-06-08:
Hello,

After updating Cura to the 15.04 version, all my previous settings were gone.
(it would be a nice feature to retrieve print settings/preferences after updating)

So I went into the settings to reset my printing preferences, and I selected particularly in "black magic" the "spiralize the outer contour*" option.

From there, as described in the picture, I had no more infill, unless I put 100%.

Once this option unchecked, I returned on a traditional view, and the filling was again considered.
Is it me who misunderstood the use of this option or is it a small problem with the software?

* I tried to translate from french
[image]

BagelOrb, 2015-06-08:
I think you totally misunderstood The spiralize option. Spiralize is a printing mode which changes a lot of other settings. Its for printing a vase from a solid cylinder object for example.

closed 2015-06-08 by BagelOrb

nallath, 2015-06-08:
Also, only the firmware of the UM2Go has been changed in 15.04

francoisba, 2015-06-09:
Ok, thank you guys. I must admit Cura is totally new to me, and with features I didn't know with Makerware.. So I still have to discover !
Thank you for your help !