daid/LegacyCura #1279: Problem inluding raft

plgarcia, 2015-06-12:
On my machin and depending on the PLA I use, the raft is missing a bit of the external border at the biginning, just the time for the presure to become sufficiant in the noze.
This causes problem for the next layer that has nothing to grip on, and somtimes that material that is not at the rihgt place is pulled later by the noze that pulls the raft.
May be the this first external line could overlap for a few centimeters to ensure it is completly closed, or extrusion of the PLA could start before while the noze goes to the starting point.

plgarcia, 2015-06-21:
In fact the problem is partially caused by the end.gcode that retracts 6 mm of filament with the two lines :
G1 E-1 F300                            ;retract the filament a bit before lifting the nozzle, to release some of the pressure
G1 Z+0.5 E-5 X-20 Y-20 F{travel_speed} ;move Z up a bit and retract filament even more

The start gcode feeds only 1 mm.

G1 F200 E7              ;extrude 1mm of feed stock

BagelOrb, 2015-06-22:
I dont know what Machine you have ban an Ultimaker 2 does the retraction and priming in the firmware. You probably messed up your starting gcode and end gcode settings.

What type of machine do you have?

plgarcia, 2015-06-22:
My machine is a prusa Mendel i3.
I have 2 different computer on wich I installed Cura, and start and end gecode are the same.

BagelOrb, 2015-06-23:
Well, does the one computer generate good results while the other doesn't? Why does it matter that you have two different computers?

Did you check whether the start gcode includes the priming?

plgarcia, 2015-06-23:
The fact I have 2 computers is that one is for work and one is personal.
The one I use is my personanl computer. But I installed cura on both to make sone tests.
The point I was making is that the two computers generate the same incorrect gcode, the same start and end gcode, and I did not change anything by hand on my computer at work. It is just a reference for me a a standard installation.

As a summerize :
I installed CURA 15.02 and I selected Prusa i3 mendel a printer. And I update to 15.04. So may be this problem could not exist any longer in 15.04 and my start and end gcode are inherited from the initial installation.

In the start.gcode the priming is :
G1 F200 E3 ;extrude 3mm of feed stock

As in end.gcode  the 2 lines follwing retract the filament for 6mm :
G1 E-1 F300 ;retract the filament a bit before lifting the nozzle, to release some of the pressure
G1 Z+0.5 E-5 X-20 Y-20 F{travel_speed} ;move Z up a bit and retract filament even more

In this situation it is obvious that the next print will miss the start for 3 mm of filament.
So I changed the 

I changed the priming to put 7 mm what is far better, and solved my problem.

I beleive cura should change that.

If you want me to make a test, as de-installing Cura, cleaning the personal data, and restart everything from scratch I can do it.
Regards