daid/LegacyCura #1280: Stopping fan

plgarcia, 2015-06-12:
For my prussa, I have not foung a way to parameterize CURA for the fan not to stop.
I have included in start GCODE this line that starts the fan : 
       M106    S255   ; Starts the fan full speed

And this works fine  but, jus after the rafts there  is always, whateve the paraméters I have chosen  a M107 generated what I would not like.
      ;LAYER:0
       M107
If min and max speed fan are identical in the configuration, no M106 is generated later so the fan stays off.

BagelOrb, 2015-06-12:
The raft has an option raft_base_fan_speed, but I don't know if it was there in the old version you are probably running.

I can't imagine this problem still exists in the current code. The fan speed code has been reworked in the mean time.

closed 2015-06-12 by BagelOrb

plgarcia, 2015-06-12:
Is the 15.04 not the last version ?
If so I will executre the current version on git

BagelOrb, 2015-06-12:
There is a 15.06, but it has a totally new front-end. For the front-end you would need Ultimaker/Cura, Ultimaker/Uranium and Ultimaker/libArcus.

They all have a 15.06 branch.