daid/LegacyCura #1286: mingw32-make: Leaving directory 'c:/Cura/CuraEngine'

maktub8, 2015-06-17:
Hi daid,
I installed MinGW,and i dont know what is wrong with it.
here is ,run in git bush,win7,x64
mingw32-make: Entering directory 'c:/Cura/CuraEngine'
mingw32-make: *** No targets specified and no makefile found. Stop.
mingw32-make: Leaving directory 'c:/Cura/CuraEngine'

Please help me,thanks.
[image]

BagelOrb, 2015-06-17:
You say you installed MinGW, but it looks as if you're trying to install Cura. What process are we seeing here? Is this perhaps the installer?

maktub8, 2015-06-17:
Thanks @BagelOrb .
You are right,I am building Cura.In package.sh 
[image]
shows me it wants to build CuraEngine,and before here runs well.
I dont know why is it.Should I reinstall MinGW?

daid, 2015-06-17:
Most likely it's trying to build the wrong CuraEngine branch.

maktub8, 2015-06-17:
@daid 
what's that mean?
and now I compile CuraEngine alone and follow this
https://github.com/Ultimaker/libArcus           
README

I'll give the result tomorrow..
Thanks daid.Good night.

rmelbourne, 2015-07-28:
@maktub8 were you able to solve this problem? I just ran into the same issue!

maktub8, 2015-07-28:
@rmelbourne sorry about you,my win7 system cant build it well.I play cura with Ubuntu now.I also very helpless.