daid/LegacyCura #1288: CURA support material display- Feature enhancement request

PJParish, 2015-06-21:
It appears that the support material lines and grids generated for overhangs are determined from a fixed reference point per the settings entered under the expert settings. They generate in the same relative place regardless of the position of the object.  Depending on the settings and the placement of the object on the bed the support structure might or might not line up with critical edges especially ones that are parallel to them when it gets displayed.  It takes a lot of tweaking and many times generating the layers view to get an object in the right position on the bed to match overhang edges with a line of support material so as to not get sagging on a long parallel edge during the print.

Enhancement:  When support material is turned on it would be really nice to see a ghost image of where the support grid or lines will be generated in the normal viewing mode on the entire grid, or at least around an object out to the outline distance, to make optimizing the placement of the objects edges with the optimized supports easier.  A layer slider that would allow the grid/line image to be moved up and down through an object would make it very versatile.

BagelOrb, 2015-06-21:
In the old cura there is the later view which does that. In the new cura there is the zigzag support which better supports critical overhangs.

I don't think many people tweak support in that way so I don't think the need is high for such functionality.

It is interesting though what you think are critical overhangs. Perhaps the support system can automatically detect and improve on such overhangs. In what cases would you call an overhang critical?

PJParish, 2015-06-21:
Thank you for the feedback.

Yes, the Layer view shows it in the current version of Cura.   Unless I'm missing something I can't change the object position in that view.  Is there a newer version of Cura available?  When I check for updates I get a message that I'm running the latest.

How do others tweak the support?  I adjust the settings on angle, fill%, and distance from object as well as the object placement on the bed to get the supports to go where I need them.  I'm not aware of another method.

Critical overhangs are those that are not bridged and the outer edge is parallel to a support line but offset by about 1 to 1.5 mm or or more or sometimes just several passes of the printing head trying to build a cantilever end, point, or loop.  You get saggy strings dipping down for several layers until it build enough support to flatten out.  Also fine details such as sharper or small areas that support larger areas above.  If those fail to center on a support structure they sag down and do not support the structure above for several layers.  They are features that are beyond the capability of cooling to keep rigid.  The resulting print requires a lot of clean up.

I avoid these occurrences as much as I can by optimizing the density of the support material structure and moving the object around on the bed to force support material to go under those points.  Rotating the object is sometimes required as well.  It just gets very tedious switching back and forth from Normal view where I can move the object around and Layers view where I can see where the support structures ended up.

Given that the support structure seems to stay fixed under an overhang area regardless of the position of the object I figured a ghost view of it in Normal view would be the easiest way to enhance my work method.  Maybe there are other options?

Having the system detect critical features automatically would be nice if it could catch floating points like printing an inverted J feature or candy cane. The teeth tips of a gear on a spindle (thus pointy overhangs floating above the bed and requiring support under each one) is an another example.  

The ability to drag around the location of specific support features would be another approach to support optimization but that seems quite complicated.  For example a selection and drag positioning option of scalable grids, lines, cones, and towers (a round hoop support structure) would be very nice to support radial arrays with fine details and overhangs.

PJParish, 2015-06-21:
Wow, that came out really long.  :)

BagelOrb, 2015-06-22:
Interactive support:
User interaction for support is on the map, but still in the distant future. I do think it would then also be nice to be able to shift the support lines and rotate them in some kind of support editor.

Critical overhangs:
These are the parts within an overhang area with worst angle. Generally the outer edges in an overhang area have highest slope. Therefore the inward corners in support areas ( = outward corners in layer parts) can be marked to be treated differently.
This will not work for hi-poly organically shaped models however, since then there would be too many critical overhangs.
If I would implement this in the future, I should make it a toggleable setting.

PJParish, 2015-07-08:
Here are some other features that generate critical overhangs.  Also viewed from below.  This part is only to show examples of various clutch, gear, and locking geometries but also demonstrates the difficulty of getting the current support structure to support all the critical edges and overhangs properly at the same time.

[image]

Parts with these types of features repeated many times become problematic to get a good support structure positioned.  The only other alternative I see is hand constructed support features to be included with the file.

PJParish, 2015-07-08:
I'm still curious how others manipulate the support structures in Cura to get precise prints.