daid/LegacyCura #1289: cura has a big space between

pedro123a, 2015-06-21:
im trying to print a dragon but is complicate because Cura has rare configuration is new for me.
the software give great space between of the body and the head,
[image]
[image]
[image]

daid, 2015-06-22:
Most likely the model has some holes in it which confuse Cura. (or the rendering messed up, as this happens sometimes)

closed 2015-06-22 by daid

BagelOrb, 2015-06-22:
Try out the x-ray view!