daid/LegacyCura #1290: Error generation gcode

plgarcia, 2015-06-22:
I encountered a problem with wrong generation of gcode with Cura 15.04
The gcode is incorrect that there is an extra face generated in the conduit.
See nozzle.stl and Nozzle.gcode here
https://drive.google.com/folderview?id=0B352dB2b8061UVlJSFRNVGJ1MlE&usp=sharing

I reported this bug at
https://github.com/Ultimaker/CuraEngine/issues/216

daid, 2015-06-22:
Error in model:
http://i.imgur.com/TDS2eKM.png

closed 2015-06-22 by daid

plgarcia, 2015-06-22:
If there is an error it is an error in FreeCad.
Are the red things errors?

nallath, 2015-06-22:
If you view the model in x-ray mode, all error areas are highlighted in red.

plgarcia, 2015-06-22:
OK thanks, I have to analyse a biut further to determine where the error is either in FreeCad, or in Cura.I let you know

daid, 2015-06-22:
If freeCAD exported that, then there is an error in FreeCAD. Solid models should never show red in X-Ray view in Cura.

plgarcia, 2015-06-22:
May be, I will analyse the object and try to state where the problem can be, and I let you know. There are two sofwares involved FreeCad and Cura, he proble can com from one or the other. IIt seemes that there is a better chance it comes from FreeCAD but that can come from Cura.

plgarcia, 2015-06-22:
Ok found,
Filet added lead to have opisite faces crossing. That is not a problem with a software but a problem my construction.

I will remind the tric  whith the red errors.
I Think we can close this one.
Thanks and sorry.