daid/LegacyCura #1292: issue with overhang angle support for cura 15.02.1

albertico931, 2015-06-25:
There is an issue in cura 15.02.1 when I change the overhange angle support to 90 degrees (Horizontal) it keeps making support in angled Area.

BagelOrb, 2015-06-29:
The support system has been totally revised. However, setting the overhang angle to 90 will not be a good idea. The overhang angle delimits what can be printed without support. Setting it to horizontal would mean that horizontal can be printed, so no support should be generated. 

However, the new Cura might crash because of the illogical value. Try setting it at 89 degree.

albertico931, 2015-06-29:
Thank you,

Jose Alberto Valle Lopez
*Director General*
alberto.valle@sketch3dmx.com
Cel: (686) 161-3153

BagelOrb, 2015-07-03:
The bug still exists: it should either be impossible to enter 90 into that field, or part of the support generation should be skipped.