daid/LegacyCura #1293: Will not print ???

rockendj, 2015-06-26:
I have a Reprap Prusa I3 After days putting it together installed software selected a print item then nothing, plate is heated & nozzle seems ok to any ideas ready to scrap it

nallath, 2015-06-26:
Sorry, we're not a Prusa helpdesk. Try the RepRap IRC.

closed 2015-06-26 by nallath

rockendj, 2015-06-26:
thanks I clicked on help / report a problem & it took me here maybe that is the problem...

daid, 2015-06-27:
This is the Cura software issue tracker.

Don't know where you bought your printer. But we're from Ultimaker. So we support our software and our printers. But we cannot support anything else.

rockendj, 2015-06-27:
My Printer came with Cura Software & there is a setting for my printer I don't understand & since "report an issue" took me here I am very confused

daid, 2015-06-27:
You'r printer supplier is what we call a "freeloader". Cura is available for free for everyone, they just ship it with the printer. Does not mean we (Cura developers) will support your printer.