daid/LegacyCura #1299: Visible 3 steps in part of larger prints - very strange

moolood112, 2015-06-29:
I need help with a strange behavior with my new Printrbot maker kit which works great except for this particular issue with Cura( I think) 
Two larger objects I have resized or rotated and printed using Cura 15.04 show 3 visible "steps" or X offset when printed. When i print the object twice, the 3 steps issue happens again in the same place (just to make sure it wasn't a random accident)
[image]
[image]

On the ears of the gremlin object (see attached)  pulled from https://www.thingiversedotcom/thing:767713
On the knee and butt of the Rodin thinker (see attached) pulled from http://www.thingiversedotcom/thing:34343

I'm thinking this is a Cura bug more than anything else but I didn't find any sign of it in the forums or the Cura github. Toughts?

milkok, 2015-06-30:
This is the problem of hardware.

moolood112, 2015-06-30:
Thank you, you are correct. Turns out the printrbot firmware had the wrong
setting for X for centering on the bed

-- 

Best Regards - Cordialement,

[image: outscale_logo] <https://www.outscale.com/>
EUROPE / NORTH AMERICA / ASIA

Stephane  Maarek
Vice President North America
Tel : +1 973 768 3658

www.outscale.us
[image: logo_facebook] <https://www.facebook.com/outscaleinc/> [image:
logo_twitter] <https://twitter.com/outscaleINC> [image: logo_google+]
<https://plus.google.com/+OutscaleUS/posts>
------------------------------
IMPORTANT: The information contained in this message may be privileged and
confidential and protected from disclosure. If the reader of this message
is not the intended recipient, or an employee or agent responsible for
delivering this message to the intended recipient, you are hereby notified
that any dissemination, distribution or copying of this communication is
strictly prohibited. If you have received this communication in error,
please notify us immediately by replying to the message and deleting it
from your computer.

closed 2015-06-30 by daid