daid/LegacyCura #1306: Cura 15.02.1 stops calculating slices after resize.

RuanSerfontein, 2015-07-07:
I am running Windows 8.1(64 bit) on a Samsung laptop. I have an Intel i3-3110 CPU and 8GB of RAM.

I have recently been experiencing problems where Cura would stop processing the .stl file after i resized the file.

I downloaded and installed the new version of Cura (15.06.1) and the problem seems to have arised after installing it. i tried uninstalling it but without luck. Cura also freezes after a while.

nallath, 2015-07-07:
So is the problem with 15.02.1 or with 15.06.1?

RuanSerfontein, 2015-07-07:
The problem is with 15.02.1. The problem arose after i installed 15.06.1.

daid, 2015-07-13:
How large is the file? (best in number of triangles)

What other software is running? Especially virus scanners and firewalls have known to mess Cura up sometimes.