daid/LegacyCura #1308: Issues installing on Ubuntu

adalast, 2015-07-10:
I am struggling with the second to last step on the install of Cura on Ubuntu.  I get the following messages in the terminal:

./package.sh debian_amd64 <-the command
Cloning https://github.com/Ultimaker/Marlin.git into _UltimakerMarlin with push URL git@github.com:Ultimaker/Marlin.git
Removing Marlin/_UltimakerMarlin_250000/
HEAD is now at c3376c7 Merge pull request #11 from AmedeeBulle/Marlin_v1_Umlauts
Already up-to-date.
/home/martin/Cura
Alreaedy on 'Marlin_v1'
Your branch is up-to-date with 'origin/Marlin_v1'.
Already up-to-date.
make: *** No rule to make target '_UltimakerMarlin_250000/wiring.o', needed by '_UltimakerMarlin_250000/Marlin.elf'. Stop.
make: *** Waiting for unfinished jobs....

I am new to Ubuntu, so I am not sure what any of the errors mean.  I am pretty sure that I get up-to-date, but not the ramifications in this case.  Can anyone tell me what is going on?

daid, 2015-07-10:
If you want to install legacy Cura, it's much easier to grab the deb from:
http://software.ultimaker.com/current/

adalast, 2015-07-10:
I try to install it and it gives me an error about a later version being installed.  I am obscenely new to Ubuntu, so I am not sure how to open software from the software center or uninstall it.  Can you point me towards a tutorial for how to even open Cura or other similar software?