daid/LegacyCura #1309: Black window and crash

plasmasolutions, 2015-07-12:
Hi Daid,

I installed the newest version from the ultimaker repository (the QT based one) and got a black screen on start. It seems that QT is not capable of creating shaders on my machine.
Specs:

- 2*Gtx560ti OC 2GB Ram,
- Intel i7
- Ubuntu 64, 15.04
- Nvidia prop drivers

Blender and any other 3D app is running flawlessly here. A log if the issue (started from the terminal):

cura_app.py 
2015-07-12 09:29:02,682 - INFO - Loaded plugin ConsoleLogger
2015-07-12 09:29:02,723 - INFO - Loaded plugin LocalFileStorage
2015-07-12 09:29:02,724 - INFO - Loaded plugin MeshView
2015-07-12 09:29:02,725 - INFO - Loaded plugin WireframeView
2015-07-12 09:29:02,726 - INFO - Loaded plugin LayerView
2015-07-12 09:29:02,726 - INFO - Loaded plugin STLReader
2015-07-12 09:29:02,727 - INFO - Loaded plugin OBJReader
2015-07-12 09:29:02,728 - INFO - Loaded plugin STLWriter
2015-07-12 09:29:02,728 - INFO - Loaded plugin OBJWriter
2015-07-12 09:29:02,729 - INFO - Loaded plugin GCodeWriter
2015-07-12 09:29:02,730 - INFO - Loaded plugin ScaleTool
2015-07-12 09:29:02,731 - INFO - Loaded plugin TranslateTool
2015-07-12 09:29:02,732 - INFO - Loaded plugin SelectionTool
2015-07-12 09:29:02,734 - INFO - Loaded plugin MirrorTool
2015-07-12 09:29:02,734 - INFO - Loaded plugin CameraTool
2015-07-12 09:29:02,742 - INFO - Loaded plugin RotateTool
2015-07-12 09:29:02,743 - INFO - Checking for new version of cura
2015-07-12 09:29:02,743 - INFO - Loaded plugin UpdateChecker
2015-07-12 09:29:02,746 - INFO - Loaded plugin USBPrinting
2015-07-12 09:29:02,747 - INFO - Loaded plugin CuraEngineBackend
2015-07-12 09:29:02,776 - WARNING - Could not find any translations matching default for file uranium_qt, falling back to english
2015-07-12 09:29:02,777 - WARNING - Could not find English translations for file uranium_qt. Switching to developer english.
2015-07-12 09:29:02,777 - WARNING - Could not find any translations matching default for file cura_qt, falling back to english
2015-07-12 09:29:02,777 - WARNING - Could not find English translations for file cura_qt. Switching to developer english.
2015-07-12 09:29:03,143 - DEBUG - Loaded theme /usr/share/cura/resources/themes/cura
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:146: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Controls/Menu.qml:146: TypeError: Cannot read property '__contentItem' of null
file:///usr/share/cura/resources/qml/SidebarSimple.qml:119:22: Unable to assign [undefined] to bool
2015-07-12 09:29:03,276 - INFO - Started engine process: /usr/bin/CuraEngine
QOpenGLShader: could not create shader
QOpenGLShaderProgram: could not create shader program
QOpenGLShader: could not create shader
QOpenGLShaderProgram::uniformLocation( imageTexture ): shader program is not linked
QOpenGLShader: could not create shader
QOpenGLShaderProgram: could not create shader program
QOpenGLShader: could not create shader
QOpenGLShaderProgram::uniformLocation( imageTexture ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( imageTexture ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( imageTexture ): shader program is not linked
QOpenGLShaderProgram: could not create shader program
QOpenGLShader: could not create shader
QOpenGLShaderProgram::uniformLocation( u_ambientColor ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( u_diffuseColor ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( u_specularColor ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( u_shininess ): shader program is not linked
QOpenGLShaderProgram: could not create shader program
QOpenGLShader: could not create shader
QOpenGLShaderProgram: could not create shader program
QOpenGLShader: could not create shader
QOpenGLShaderProgram: could not create shader program
QOpenGLShader: could not create shader
QOpenGLShaderProgram: could not create shader program
QOpenGLShader: could not create shader
QOpenGLShaderProgram::uniformLocation( u_ambientColor ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( u_diffuseColor ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( u_specularColor ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( u_overhangColor ): shader program is not linked
[... 121 lines cut ...]
QOpenGLShaderProgram::uniformLocation( textureScale ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( dpr ): shader program is not linked
QOpenGLShaderProgram: could not create shader program
QOpenGLShader: could not create shader
QOpenGLShader: could not create shader
shader compilation failed: 
""
QOpenGLShaderProgram::uniformLocation( matrix ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( color ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( textureScale ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( dpr ): shader program is not linked
QOpenGLShaderProgram: could not create shader program
QOpenGLShader: could not create shader
QQuickCustomMaterialShader: Shader compilation failed:
""
QOpenGLShader: could not create shader
QOpenGLShader: could not create shader
QOpenGLShaderProgram::uniformLocation( qt_Matrix ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( src ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( qt_SubRect_src ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( color ): shader program is not linked
QOpenGLShaderProgram::uniformLocation( qt_Opacity ): shader program is not linked
Speicherzugriffsfehler (Speicherabzug geschrieben)

Greetings, Thomas

nallath, 2015-07-13:
This is not the repo for the version with QT. Please post the issue there.

closed 2015-07-13 by nallath

plasmasolutions, 2015-07-13:
Hi Nallath, could you lead me to the right repo? I couldn't find one...

nallath, 2015-07-13:
https://github.com/Ultimaker/cura