daid/LegacyCura #1313: Cura seems to add elements that are not part of the model.

ErwinRieger, 2015-07-16:
I sliced this model: http://ibrieger.de/public/halter_karmann_rechts.stl with cura 15.04 and there are  parts appearing in the sliced object that are not part of the original model.

In cura it is visible in the layers view but not in any of the other views, xray view shows no problems.

The model is created by openscad.

In the image below you see the additional parts at the lower edge. The problematic section is between the two holes and on the right of the right hole. There you can see additional material on the outside of the model. 

The printed object has this additional material also.

[image]

The profile used:

---------------------------------------------------------------------------------------------------------------
[profile]
layer_height = 0.2
wall_thickness = 0.8
retraction_enable = True
solid_layer_thickness = .6
fill_density = 30
nozzle_size = 0.4
print_speed = 75
print_temperature = 220
print_temperature2 = 0
print_temperature3 = 0
print_temperature4 = 0
print_temperature5 = 0
print_bed_temperature = 70
support = None
platform_adhesion = None
support_dual_extrusion = Both
wipe_tower = False
wipe_tower_volume = 15
ooze_shield = False
filament_diameter = 2.85
filament_diameter2 = 0
filament_diameter3 = 0
filament_diameter4 = 0
filament_diameter5 = 0
filament_flow = 100.0
retraction_speed = 40.0
retraction_amount = 4.5
retraction_dual_amount = 16.5
retraction_min_travel = 1.5
retraction_combing = All
retraction_minimal_extrusion = 0.02
retraction_hop = 0.0
bottom_thickness = 0.3
layer0_width_factor = 110
object_sink = 0.0
overlap_dual = 0.15
travel_speed = 150
bottom_layer_speed = 15
infill_speed = 85
solidarea_speed = 50
inset0_speed = 0.0
insetx_speed = 0.0
cool_min_layer_time = 5
fan_enabled = True
skirt_line_count = 1
skirt_gap = 3.0
skirt_minimal_length = 150.0
fan_full_height = 0.5
fan_speed = 100
fan_speed_max = 100
cool_min_feedrate = 10
cool_head_lift = False
solid_top = True
solid_bottom = True
[... 179 lines cut ...]
	M117 Printing...
end4.gcode = ;End GCode
	M104 T0 S0                     ;extruder heater off
	M104 T1 S0                     ;extruder heater off
	M104 T2 S0                     ;extruder heater off
	M104 T3 S0                     ;extruder heater off
	M140 S0                     ;heated bed heater off (if you have it)
	G91                                    ;relative positioning
	G1 E-1 F300                            ;retract the filament a bit before lifting the nozzle, to release some of the pressure
	G1 Z+0.5 E-5 X-20 Y-20 F{travel_speed} ;move Z up a bit and retract filament even more
	G28 X0 Y0                              ;move X/Y to min endstops, so the head is out of the way
	M84                         ;steppers off
	G90                         ;absolute positioning
	;{profile_string}
support_start.gcode = 
support_end.gcode = 
cool_start.gcode = 
cool_end.gcode = 
replace.csv = 
preswitchextruder.gcode = ;Switch between the current extruder and the next extruder, when printing with multiple extruders.
	;This code is added before the T(n)
postswitchextruder.gcode = ;Switch between the current extruder and the next extruder, when printing with multiple extruders.
	;This code is added after the T(n)

---------------------------------------------------------------------------------------------------------------

BagelOrb, 2015-07-17:
Try 15.06 its improved a lot!!!! BTW report problems of that version at Ultimaker/Cura

ErwinRieger, 2015-07-17:
15.06 does not work on my debian jessie system - pyqt version 5.3.2 is too old for new cura :-(

BTW should i file this issue at Ultimaker/Cura?

BagelOrb, 2015-07-19:
Well as of yet, this issue only seems to occur in the old Cura. Since the new Cura uses (a quite rigorously updated version of) the same CuraEngine, the problem is likely to exist in the new Cura as well.

I will slice your model tomorrow to see if I can reproduce the bug in the new Cura.

As to your PyQt, shouldn't you just update it to a newer version then?

When reporting an issue, please supply info on your OS and Cura version right away.

ErwinRieger, 2015-07-19:
Thanks for trying with the new Cura :-)

PyQt 5.3.2 IS the newest available version for Debian stable/jessie, and I don't want to update or mix my system with debian testing or debian unstable...

BagelOrb, 2015-07-20:
The problem still exists in the new Cura :(
I find this bug boggles me.
I think it must be a bug in the underlying polygon library, but I'd have to check.

BagelOrb, 2015-07-20:
The problem has been solved in the new Cura. The solved stuff will be in the next release.

closed 2015-07-20 by BagelOrb