daid/LegacyCura #1314: Cura doesn't print very last part of surface as shown picture

3dprotech, 2015-07-19:
Dear,

Whenever print something that has flat surface, very last part of surface was not printed.
If I print 6 pieces at once, only very last one shows above symptom.
I need your support to resolve this issue.

Thank you for your help and contribution !!
[image]

Dim3nsioneer, 2015-07-19:
Is the missing part displayed in Cura's layer view? If you have troubles to see it there, you may check at http://gcode.ws or use Repetier host or any other gcode viewer.

nophead, 2015-07-19:
If you are using Marlin firmware you probably need to put M400 at the start of your end gcode to wait for the buffered moves to finish before turning things off. Marlin buffers moves but executes any M codes following them immediately, i.e. out of order and that can cause problems like this.

daid, 2015-07-19:
Yes, this isn't a Cura specific problem, it's a printer/setup specific problem. So closing it.

closed 2015-07-19 by daid

3dprotech, 2015-07-21:
Thanks for quick & kind support.
Youe solution works fine as attached image.
[image]