daid/LegacyCura #1315: Save model does not honor scale changes

CountMurphy, 2015-07-19:
I have tested this with both the Windows and Linux clients. I have a model which I accidentally scaled incorrectly when I created it. The model in question is 76.4 mm on the x axis. My typical workflow is to load the model into cura to position, rotate and scale it 76.7 mm, save the stl and then send it to replicatorg to export to my makerbot. With the new release of  cura, any model I save as an stl retains the models original dimensions. For example I can import my model, scale to 76.7 mm (on the x axis), save the model to stl and open that newly saved model with cura. The loaded model will still have the dimensions of 76.4.

nallath, 2015-07-21:
This is a 15.06 bug, so the correct repo is https://github.com/Ultimaker/cura

closed 2015-07-21 by nallath

CountMurphy, 2015-07-21:
My apologies. Thanks for pointing me in the correct direction