daid/LegacyCura #1316: Issue trying to package Cura

rmelbourne, 2015-07-20:
I am trying to create a new cura release for Cura 15.04 on Windows and I am using the command "package.sh" on git bash for Windows 7 32 bit and it is giving me the following error:

[image]

I know this means that the directory is non-existent or empty but I am not sure how to solve this. It seems like nothing is being extraced from PortablePython because the log.txt file prints out the following:
 -------------------------------
7-Zip [32] 15.05 beta : Copyright (c) 1999-2015 Igor Pavlov : 2015-06-14

Scanning the drive for archives:
1 file, 236381640 bytes (226 MiB)

Extracting archive: PortablePython_2.7.6.1.exe
--
Path = PortablePython_2.7.6.1.exe
Type = Nsis
Physical Size = 236381640
Method = LZMA:25
Solid = +
Headers Size = 2263106
Embedded Stub Size = 66048
SubType = NSIS-Park-3 Unicode log

No files to process
Everything is Ok

Files: 0
Size: 0
Compressed: 236381640
7z x -y PortablePython_2.7.6.1.exe $_OUTDIR/Lib/site-packages

------------------------------
Any ideas on what could be happening?

daid, 2015-07-20:
Is the PortablePython file correctly downloaded? The file should be about 46MB

rmelbourne, 2015-07-21:
The size is as you said it should be

[image]

I think these lines in "package.sh" are the ones that are not working:

630    extract PortablePython_${WIN_PORTABLE_PY_VERSION}.exe \$_OUTDIR/App
631    extract PortablePython_${WIN_PORTABLE_PY_VERSION}.exe \$_OUTDIR/Lib/site-packages

3dht, 2015-07-28:
Hi @rmelbourne  and @daid , i have exactly the same error. Actually i read this script package.sh and i didn't find out variable $OUTDIR before this function , so for sure program cannot use it to extract our file PortablePython. Do u guys know its destination path ? with this we can precise absolute path in this function .

I'm waiting your replies

Sincerely,

daid, 2015-07-28:
There is no $OUTDIR variable, that's actually part of the output path. Hench the blackslash. (Exe installers are odd like that)

3dht, 2015-07-28:
@daid: ah ok thank you for your information , however do you know why do we cannot extract this file ? Because though i finished command extract extract "PortablePython_${WIN_PORTABLE_PY_VERSION}.exe \$_OUTDIR/App
extract PortablePython_${WIN_PORTABLE_PY_VERSION}.exe \$_OUTDIR/Lib/site-packages"
i didn't find out folder App or Lib in main folder :(
Thank you in advance and i'm waiting your reply

Regards,

maktub8, 2015-10-27:
@3dht  @rmelbourne Hi guys，have you solved it? @daid I have the same error,please help..

donaldding, 2016-03-24:
I got this problem,  and I delete the PortablePython.exe file, and run the script again then the problem is gone. Maybe the file was not correctly downloaded before.   I'm on Mac