daid/LegacyCura #1324: 15.06.03 Raft Not Displaying in Layers View

Fuzionist1, 2015-08-03:
The raft is now displaying in Layers view.  Instead, the first layer shown is as if there was no raft at all. I have tried complex and completely basic models. 
[image]

BagelOrb, 2015-08-04:
In the old Cura the outputted gcode was parsed and lines were thus displayed. In the new Cura each feature (walls, infill, etc.) is sent from the backend separately and viewed separately.

The raft was probably forgotten. 

Moreover, the layerview is an indication as to what is exactly printed. The actual gcode may diverge from the view.

BagelOrb, 2015-08-04:
Please submit this issue in the right repository:
https://github.com/Ultimaker/Cura

closed 2015-08-04 by BagelOrb