daid/LegacyCura #1328: usability: user doesn't realize you can rotate models

martinxyz, 2015-08-07:
Someone in a fablab just told me he went to great lengths to rotate a model by 180 degees outside of Cura. Why did he not realize that Cura can do this? Turns out that Cura did not show any rotate buttons because the model was not selected.

Possible solutions could be to always keep at least one model selected if there is one, or to always show the buttons but gray them out.

daid, 2015-08-07:
15.06 always shows the rotation tools.

closed 2015-08-07 by daid

martinxyz, 2015-08-07:
I tested with the latest git, SteamEngine branch. There is no 15.06 tag in this repository, maybe you forgot to push? This version still hides the rotation tools as soon as I click somewhere. But it shows the tools on the first model load, and if it didn't do that before then yes, this should have solved the issue (got no fresh user at hand to actually verify :-)

daid, 2015-08-07:
Development moved to https://github.com/Ultimaker/Cura