daid/LegacyCura #1330: When printing objects with walls that are a multiple of the nozzle size infill is still produced.

lawrencejohnston, 2015-08-14:
I'm seeing this in Cura 15.04.02. My profile is included in the Gcode linked below.

When I'm printing objects with thin walls that are a multiple of the nozzle size that should produce no infill (in this case 1.6mm walls with a 0.4mm nozzle and the shell set to 0.8mm) sometimes I end up with infill anyway and I can't figure out why. It's a problem both because of the extra print time and because for thinner walls (e.g. 0.8mm) it results in a wavy wall because the nozzle zigzags back and forth for the infill.

Here's an example that reproduces the issue: [Scad, STL, and Gcode](https://gist.github.com/lawrencejohnston/4b368f7d8e306f2487cc)

Search for "FILL" and you'll see where it's doing infill that it shouldn't. Looking at the Gcode it looks like one of the walls is being made 16.001mm thick resulting in the infill.

It seems to happen more often when I slice multiple objects on the same plate, but I've seen it with single objects as well (though that seems harder to reproduce for some reason).

What's going on here?

BagelOrb, 2015-08-14:
There is no occurrence of "TYPE:INFILL" in that gcode.

Try to set Infill to 0% and set Fill Gaps Between Walls to False

BagelOrb, 2015-08-14:
Also, try the new Cura!
This repo is of the old Cura. The new Cura's repo is at https://github.com/Ultimaker/Cura

lawrencejohnston, 2015-08-14:
There are lots of occurrences of infill in that Gcode. Search for `;TYPE:FILL`. Line 118 is an example. Please reopen this issue.

Setting infill to 0% is not a good workaround because the actual models I'm printing have parts that do need infill (for example the base and the latch. Setting fill between gaps is a possibility although it obviously negatively affects the aforementioned areas as well.

I tried the new Cura a month or two ago and ran into some problems/missing features that sent me back to the old one. I'll try it again at some point but for now I'd just like to get my stuff printed.

BagelOrb, 2015-08-17:
Oh I'm sorry. I searched the wrong term.

I think another work-around would be to set the infill-overlap to 0%, for now.

I will see whether this indeed causes the problem and solve the issue if it is present.

BagelOrb, 2015-08-17:
The issue is solved now. The next release will not include the bug.

closed 2015-08-17 by BagelOrb

lawrencejohnston, 2015-08-17:
Thanks Tim. I appreciate the quick fix.