daid/LegacyCura #1332: wtf is wrong with CURA 15.06.03?

erikstiff, 2015-08-20:
I Install the new cura today, just because the other one (15.04) refuse to open my stl. Don't know why the old one did that, but decide to install the new version.
The new version looks great, BUT, I can't rotate or move my object. There is no place to put the price for your filament, so you can know the cost of the object. That was very important for me!
Most of the times when I close CURA he generates a error log. I'm using the older version again (and now it opens stl)
Can someone help me? The new version looks better, but I can't do anything on it!

nallath, 2015-08-20:
You posted the issue in the wrong repository. The new cura can be found at https://github.com/Ultimaker/cura

The new Cura has a plugin that allows you to calculate the price, which can be found here; https://github.com/nallath/PrintCostCalculator

Simply stating that cura generated an error log does not help us tracking down the issue. Post the log.

daid, 2015-08-22:
And, there is nothing wrong with it, we just re-build it from scratch. Which means some things are still under construction or forgotten.