daid/LegacyCura #1337: First layer(s) temperature

kengdahl, 2015-09-15:
First of all I'm sorry if this is a clone of an existing request. I could not find one when I searched. 

To make it short and sweet:
I'm missing the option to print the first layer(s) hotter than the rest in Cura.

tzminversion, 2015-09-15:
Yes I would also like to see this option! The first layer is always the tricky one and to print at a higher temp would stick better for the first layer.

nallath, 2015-09-15:
tweak at z

closed 2015-09-15 by nallath

kengdahl, 2015-09-15:
I should have added that I'm using Cura 15.06.03. It seems like the new beta version does not support the old plugins or have the option to change temperatures at a given layer height.

So I don't see the reason to close this issue.

nallath, 2015-09-15:
This is not the repo for 15.06. This repo is for 15.04 and below, which supports this behaviour.

15.09.81 supports this behaviour through the PostProcessing plugin which is shipped with Cura. You can find the plugin in one of my repos. The latest beta of cura can be found at http://software.ultimaker.com/Cura_closed_beta/15.10/