daid/LegacyCura #1339: Cura 15.04 Dual Extrusion mis-align

XJoonK, 2015-09-23:
Every time I import two models (one for support) and select "Dual Extrusion Merge" there is always a slight mis-alignment with the two pieces, even though they were imported from the same file, and should be aligned.

I looked for this issue and did not see it addressed already.  I apologize if the answer is obvious, I am still a novice.

-XJoonK
[image]

daid, 2015-09-24:
Then the origins of the export are not the same from AutoCAD. So this problem lays more with the AutoCAD exports then Cura.

XJoonK, 2015-09-24:
Thank you for your response.  
If this is not a common problem, then I am inclined to agree that it is an AutoCAD export problem.  I will try exporting through Blender and see if the problem persists.

XJoonK, 2015-09-24:
The problem occurs in MatterControl as well, so it is an AutoCAD problem.

XJoonK, 2015-09-24:
So

XJoonK, 2015-09-24:
So there is no way to manually align these?  It's a little frustrating.  

Am I seriously the only person experiencing this kind of issue?

BagelOrb, 2015-09-28:
You are filing the bug in the repository of the old cura interface. Please file a new issue under Ultimaker/Cura

nallath, 2015-09-28:
The new Cura doesn't have this feature either. It assumes your model is correctly aligned.

XJoonK, 2015-09-28:
The issue was due to the original file having negative coordinates.  The pieces were shifting to make all  coordinates positive.  Thank you all for your help, the problem of 'use error' has been fixed for this issue.

closed 2015-09-28 by XJoonK