daid/LegacyCura #1340: Translation to Russian

akypb, 2015-09-27:
Hello!

I try to make a good localization of the Cura to Russian with aim to simplify installing process of the Cura in education process.
I never before use Python nor OpenGL, and I have no dubt that many things I did by wrong way.
But I try to fix all things in Cura code that interfere with the localization process.
Some fixes I made because in Russian we have at least two plural form of a noun, i.e. "one hour" "two hours" "five hours" "21 hours" but "odin chas" "dva chasa" "pyat chasov" and, there are the crazy thing, "21 chas", we have again the singular form. Even Google translate have problem with it.
This is screenshot of what I get in finish of the process.
[image]
[image]
It can be interesting for developers? How I can send my modifications to them?

BagelOrb, 2015-09-28:
You can send modifications by sending a pull request.

The translation into Russian should be a designated translation file, which
is generally provided by a russian reseller of Ultimaker printers.

Please make a pull request of your changes so that we can review your work.

Thanks in advance.

akypb, 2015-10-01:
Hmm. I will try to do it, but it is not so easy. I made more than hundred fixes in twenty files, add a .ttf font with cyrillic support, python image library (pillow) to grab the letter bitmap and return localization to wxpython library. And, of course, fix almost all translation error in cura.po file (more than six hundred fixes).

BagelOrb, 2015-10-01:
I don't understand why you have changed that many files. 

You should try to capsulate corresponding fixes in separate commits, because if you file a pull request with that many changes all in one commit, we will most likely have to disregard it because it will be very hard to review.

If you have made several changes which are of a different sort, you should send them in a separate pull request.

It's too bad you didn't contact us before you started changing the source for Russian support; then we could have told you the right procedures which would have provided you with a nice work flow as well.

I hope you can figure out how make a pull request(s) out of your fixes, so that everyone can benefit from your work.

akypb, 2015-10-01:
I'll try to explain.
First of all, many function output messages not trough gettext
1. All messages in validators.py, objectscene.py, printWindow.py, newVersionDialog.py
2. Some messages in preferencesDialod.py, sceneView.py, engineResultView.py, simpleMode.py (i.e. name of profiles you can't translate by standard way)
3. In some function output not worked standard way, i.e. in floating labels.
4. I found, that if you start to use gettext for localization, you lost ability to use standard GLUT function glutBitmapCharacter for text output.
This function out nothing and you lost all buttons labels on main screen. I modified openglHelpers.py to fix it.
5. And because I use different way for text output I modified openglGui.py sceneView.py too.
6. I found that old way for message output is not suitable, I don't understand why, but when I try use gettext in some messages, python did not start at all. 
That why I modify output for time/lenght/cost message through the openglGui/glGuiLayoutGrid class.
6. Add some confirmations for new users. I.e. if you select Ultimaker as main printer by mistake, you can easily rewrite firmware without any warnings from the Cura. I think, it's a bad practice.
7. add information about pillow library and .ttf font in about.py
8. And when I try to build  all i found that in new version of NSIS some function deprecated, then I modified installer.nsi 
and I still continue working with plurals form.
Repeat, I have zero expirience in python and OpenGL. All this modification I made in last two-three weeks, and I never before use github or something else.

BagelOrb, 2015-10-05:
@awhiemstra you should take a look at this. There might be some valuable work here.

@akypb You should make a pull request out of the changes you've made.