daid/LegacyCura #1341: OS X 10.11 El Capitan Cura Error

cakhoury, 2015-10-02:
Just after updating my MacBook Pro to OS X El Capitan, I cannot pan the print bed or move my object without immediately receiving the error depicted in the attached screenshot. Oddly, when zooming or changing print setting values in the side-bar, I do not receive the error. It is only when panning view or moving/rotating my object.

[image]

daid, 2015-10-02:
Problem comes from deep within a library we use:
https://github.com/wxWidgets/wxWidgets/blob/master/src/osx/core/hid.cpp#L172

nallath, 2015-10-02:
aka; Sorry, we probably cant fix this.

cakhoury, 2015-10-02:
I really appreciate your quick replies guys! Do you think it might be possible for me to disable the error somehow by editing the source code? Again, thank you so much for getting back to me so promptly!

daid, 2015-10-02:
Problem is in the code of a very complex library that we use. And we don't compile the library ourselves. Which means in theory you can edit the source of the library, compile it, insert into Cura and the message will be gone. But in practice you need to do a whole lot to actually do this. And I've never done it myself.

cakhoury, 2015-10-02:
Thank you for the information Daid. I'm very impressed with how quickly you got back to me. Hope you have a great weekend

novemberalpha, 2015-11-12:
Hey, I'm keeping an eye on this as I want to upgrade to El Capitan but won't because it kills Cura. Is this going to be resolved ever? I kinda heart your app.

daid, 2015-11-12:
It's not going to get fixed for the 15.04.x versions. Your best bet is the new betas:
http://software.ultimaker.com/Cura_closed_beta/15.10/