daid/LegacyCura #1343: X axis overshooting/getting stuck

duduabdu, 2015-10-03:
hello!
i started using my new printrbot simple metal, and after printing the 3mm Box for calibration a couple of time, i went ahead to printing the Fan Shroud.
it started well (beside the first layer, but thats another question), but just when is finished the base of the Shroud, the print head moved all the way to right(x axis) and got "stuck", tying to keep moving left and making a nasty noise.
i immediately unplug the power, but i couldn't figure out what went wrong. i uploaded the Fan Shroud file again, and on this time the printer finished the base, started to make the duct (it went like 50mm up) and did the same thing - moved all the way to right and gut stuck.
what could it be? am i doing something wrong? on the setting everything is according to the simple metal spacs (150x150x150).
My start.gcode is:

G28 X0 Y0 Z0
G92 E0
G29

BagelOrb, 2015-10-04:
Could you provide the gcode file? The original stl file would also be helpful.

duduabdu, 2015-10-04:
I am using Cura. i downloaded the "Improved Printrbot Fan Duct" from thingiverse at this link http://www.thingiverse.com/thing:356001
 if i understand you correct i uploaded the file at cura, and saved the g.code. here it is http://www.filedropper.com/fanshroudv72
 I tried 3 different models and at every one it did the same thing.
thx for the help

duduabdu, 2015-10-05:
?

BagelOrb, 2015-10-05:
It has probably something to do with USB printing.

@nallath Can you say something about this?

duduabdu, 2015-10-05:
I just tried to print a model that  i know for a fact that is not corrupted.
When i used the USB connection the print stopped at 80% of the way and crashed to the left.
I uploaded the same file to the SD card and it printed it 100% with no problems, so i can say its most likely to be a USB problem, any advice how to fix it?

nallath, 2015-10-06:
Moving to one side and getting stuck is that the head tries to move to an impossible x-axis position with the end switches disabled.

BagelOrb, 2015-10-06:
Indeed, but how is this related to USB printing?

duduabdu, 2015-10-06:
the bed moves not the head, and its moves to the other side of the end switch (at the printrbot the end switch it at X=0)

BagelOrb, 2015-10-06:
@nallath in printrbots the x moves are performed by moving the bed in the x direction.

We assumed a normal axis system, like the Ultimaker printers.

@nallath why does USB printing result in such problems?

BagelOrb, 2015-10-06:
Whoops, this is the old Cura. I shouldn't ask @nallath, you should ask @daid.

duduabdu, 2015-10-06:
i have version 15.04 , isn't that the new version ?

nallath, 2015-10-06:
Nope; 15.09.90 is the newest version (although it is in beta)

duduabdu, 2015-10-06:
i bought a high quality USB cable with  a ferrite core, i don't think the cable was the problem but i want to eliminate every option. 
i still would like to know why does USB printing result in such problems
thanx
David

nallath, 2015-10-06:
USB printing has never been very stable. It's not a focus for us as very little people with an Ultimaker print through USB (and Ultimaker does pay us ;))