daid/LegacyCura #1345: Crash on rotate

krekr, 2015-10-05:
I must say: Cura is an awesome piece of engineering; especially if you consider what we came from :)
Since I thought I'd help out the development, here's a crash report when trying to rotate 3 moderately complex STLs (1mb, 1mb, 11mb). I couldn't replicate this, so it's not really an issue, but it might help stability anyway.

Here's the OS X dump:

Process:         Cura [25693]
Path:            /Applications/Cura/Cura.app/Contents/MacOS/Cura
Identifier:      com.ultimaker.Cura-15.02.1
Version:         15.02.1 (15.02.1)
Code Type:       X86-64 (Native)
Parent Process:  launchd [313]
Responsible:     Cura [25693]
User ID:         501

Date/Time:       2015-10-05 20:14:03.931 +0200
OS Version:      Mac OS X 10.9.5 (13F1112)
Report Version:  11
Anonymous UUID:  D817D55F-287F-F108-9623-172F1E3F0CB2

Sleep/Wake UUID: 6CFB9374-84EA-447E-A83B-FEF02DCDC012

Crashed Thread:  0  Dispatch queue: com.apple.main-thread

Exception Type:  EXC_BAD_ACCESS (SIGSEGV)
Exception Codes: KERN_INVALID_ADDRESS at 0x0000000000000020

VM Regions Near 0x20:
--> 
    __TEXT                 0000000100000000-0000000100006000 [   24K] r-x/rwx SM=COW  /Applications/Cura/Cura.app/Contents/MacOS/Cura

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   org.python.python             	0x00000001039fde10 _PyObject_GC_Track + 416
1   org.python.python             	0x00000001039fea7d _PyObject_GC_Malloc + 269
2   org.python.python             	0x00000001039feb35 _PyObject_GC_New + 21
3   org.python.python             	0x000000010392e6d6 PyGen_New + 22
4   org.python.python             	0x00000001039bea3d PyEval_EvalFrameEx + 24125
5   org.python.python             	0x000000010392df75 PyGen_NeedsFinalizing + 245
6   org.python.python             	0x000000010390f6ed PySequence_Tuple + 285
7   org.python.python             	0x000000010396d2e8 _PyTuple_Resize + 776
8   org.python.python             	0x0000000103971c03 PyType_GenericAlloc + 8787
9   org.python.python             	0x0000000103909362 PyObject_Call + 98
10  org.python.python             	0x00000001039bc2c8 PyEval_EvalFrameEx + 14024
11  org.python.python             	0x00000001039c1149 PyEval_EvalCodeEx + 2137
12  org.python.python             	0x00000001039bea3d PyEval_EvalFrameEx + 24125
13  org.python.python             	0x00000001039c1149 PyEval_EvalCodeEx + 2137
14  org.python.python             	0x00000001039bea3d PyEval_EvalFrameEx + 24125
15  org.python.python             	0x00000001039c1149 PyEval_EvalCodeEx + 2137
16  org.python.python             	0x00000001039bea3d PyEval_EvalFrameEx + 24125
17  org.python.python             	0x000000010392df75 PyGen_NeedsFinalizing + 245
18  org.python.python             	0x000000010390f6ed PySequence_Tuple + 285
19  org.python.python             	0x000000010396d2e8 _PyTuple_Resize + 776
20  org.python.python             	0x0000000103971c03 PyType_GenericAlloc + 8787
21  org.python.python             	0x0000000103909362 PyObject_Call + 98
22  org.python.python             	0x00000001039bc2c8 PyEval_EvalFrameEx + 14024
23  org.python.python             	0x00000001039c1149 PyEval_EvalCodeEx + 2137
24  org.python.python             	0x000000010393a910 PyClassMethod_New + 800
25  org.python.python             	0x0000000103909362 PyObject_Call + 98
26  org.python.python             	0x00000001039b9ebd PyEval_EvalFrameEx + 4797
27  org.python.python             	0x00000001039c1149 PyEval_EvalCodeEx + 2137
28  org.python.python             	0x000000010393a910 PyClassMethod_New + 800
29  org.python.python             	0x0000000103909362 PyObject_Call + 98
30  org.python.python             	0x000000010391b97b PyInstance_NewRaw + 1099
31  org.python.python             	0x0000000103909362 PyObject_Call + 98
32  org.python.python             	0x000000010397612a _PyObject_LookupSpecial + 1386
33  org.python.python             	0x0000000103909362 PyObject_Call + 98
34  org.python.python             	0x00000001039bc2c8 PyEval_EvalFrameEx + 14024
35  org.python.python             	0x00000001039c1149 PyEval_EvalCodeEx + 2137
36  org.python.python             	0x00000001039bea3d PyEval_EvalFrameEx + 24125
37  org.python.python             	0x00000001039bf86d PyEval_EvalFrameEx + 27757
38  org.python.python             	0x00000001039c1149 PyEval_EvalCodeEx + 2137
[... 612 lines cut ...]
shared memory                        68K
===========                      =======
TOTAL                             736.2M

Model: MacBookPro8,2, BootROM MBP81.0047.B2A, 4 processors, Intel Core i7, 2 GHz, 8 GB, SMC 1.69f4
Graphics: Intel HD Graphics 3000, Intel HD Graphics 3000, Built-In, 512 MB
Graphics: AMD Radeon HD 6490M, AMD Radeon HD 6490M, PCIe, 256 MB
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1333 MHz, 0x0198, 0x393930353432382D3035312E4130304C4620
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1333 MHz, 0x0198, 0x393930353432382D3035312E4130304C4620
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0xD6), Broadcom BCM43xx 1.0 (5.106.98.100.22)
Bluetooth: Version 5.2.7f4 15858, 3 services, 15 devices, 1 incoming serial ports
Network Service: AirPort, AirPort, en1
Serial ATA Device: Crucial_CT512MX100SSD1, 512,11 GB
Serial ATA Device: TOSHIBA MK5065GSXF, 500,11 GB
USB Device: Hub
USB Device: BRCM2070 Hub
USB Device: Bluetooth USB Host Controller
USB Device: Apple Internal Keyboard / Trackpad
USB Device: FaceTime HD Camera (Built-in)
USB Device: Hub
USB Device: USB2.0 Hub
USB Device: SpaceNavigator
USB Device: MTE-450
USB Device: IR Receiver
Thunderbolt Bus: MacBook Pro, Apple Inc., 22.1

jaisongreen, 2015-11-20:
I am having this problem as well.

daid, 2015-11-21:
Nothing I can do about it. It's down to that wxWidgets is buggy as hell under OSX. So the only workaround is switching to Windows or Linux.

We have a beta for a total replacement of the frontend of Cura out. You can check it out here:
http://software.ultimaker.com/Cura_closed_beta/15.10/
As it should solve these kind of bugs. It does overhaul the whole UI, so it make take a bit to get used to.

closed 2015-11-21 by daid

krekr, 2015-11-23:
It’s not that bad actually. Thanks for getting me in the closed beta :)