daid/LegacyCura #1349: Problem printing Ultimaker Robot Sample model

ZhuDaHai, 2015-10-17:
I have a Chinese Convienent 3D Printer, which according to the instructions should be set as a Pursa Mendel I3.
I've tried printing it via USB and SD Card and at about 15% it starts freaking out and pulling the filament out about 1" worth and then jamming it back in. This total blows everything and from that point on nothing gets printed. It does this again and again at an ever quickening pace like its possessed or something.

So I killed it.

I through it might be related to the "E-Steps per 1mm filament" set to 300 - according to the manual. So I set it to 0 to use the firmware setting. Then it started not wanting to stick the filament to the brown masking tape I had on the print bed. 

Thinking I might need to recalabrate the Z 0 stop to .4mm from the head. I tried that but it made no diference. I even tried setting the "E-Steps per 1mm filament" back to 300, but it still had problems sticking the filament to the bed. 

I even tried closing all the doors and turning off all the fans in the room thinking it was cooling the filament down too fast. No luck.

Can anybody help?
Why is it freaking out on the Robot sample at 15%??
And why isn't the filament sticking to the tape anymore??

Thanks!

ZhuDaHai, 2015-10-17:
Well...
Enable Retraction was checked. 
Unchecked it enabled the Robot to be printed.
Still having a little problem getting the filament to stick to the paper tape on the bed.
Any ideas on that one?

Thanks!

BagelOrb, 2015-10-20:
Try other adhesion types.
Google on build plate adhesion.

This is not a software issue.

closed 2015-10-20 by BagelOrb